snsten/Klein #4: Missing Encoder route in PCB for Pro Micro Controllers

snsten, 2022-10-31:
### Description of Error

- There was a typo in schematics related to Pro Micro Controllers
- ENCA should have been same as ENC_A and ENCB same as ENC_B (can be seen in attached Image)
- It causes the Pro Micro based controllers not be able to use rotary encoder 
- XIAO based controllers are not affected due to the additional jumper path

[image]

### Resoultion of Error
- If you have printed the PCB already and will use wireless version with XIAO, no change is required
- If you plan to use Pro Micro based controller everthing except encoder works
- If you want to use encoder anyway you can connect wires from encoder to pins on Pro Micro

### Summary of Fix
- I have added the routes on PCB and fixed the typo in schematics
- Removed the Gerbers file until I verify the new PCB works (or if someone else try it first)

azhizhinov, 2022-11-02:
Hello.
May I ask you about a firmware, intended to use with your keyboard?
Is that pretty KLOR firmware?
If so, whether this options can be changed from these:
`/* encoder support */
#define ENCODERS_PAD_A       { F5 }
#define ENCODERS_PAD_B       { F4 }
#define ENCODERS_PAD_A_RIGHT { F4 }
#define ENCODERS_PAD_B_RIGHT { F5 }
#define ENCODER_RESOLUTION 2`
To these:
`/* encoder support */
#define ENCODERS_PAD_A       { D1 }
#define ENCODERS_PAD_B       { D0 }
#define ENCODERS_PAD_A_RIGHT { D0 }
#define ENCODERS_PAD_B_RIGHT { D1 }
#define ENCODER_RESOLUTION 2`
Is that will fix the issue for ProMicro-based?
I'm asking, because I've already ordered the PCB. :-)
Thanks for your time.

snsten, 2022-11-02:
Hi, thanks for your interest in this project.

- The firmware used with XIAO-BLE is ZMK. I have a working prototype at [Klein-zmk](https://github.com/snsten/Klein-zmk) repo.
- For Wired version with Pro Micro it will run on QMK.
- I have a test version of QMK with buzzer and touchpad working, yet to be tested are haptics, encoders and trrs connection.
- I will upload the QMK code when I have tested all the features work correctly.

As for the encoder using pin D1 and D0 should work but you will loose the ability to connect OLED or Trackpad if that's fine then you can use them by bridging the jumper JP9 and JP10.

idesignstuff, 2022-11-06:
Planning to use the Xiao BLE for mine. Ordered 10 boards from jlcpcb tonight. Thanks for all the effort to make this!

azhizhinov, 2022-11-06:
Hello.
May I ask you about batteries, you've planned to use?
How the batteries should be attached to the PCBs, that's not covered yet.
Thanks in advance.

snsten, 2022-11-07:
- The battery can be any 3.7V lipo like [these](https://www.lipobattery.us/tag/thickness-3mm/) keeping dimension and capacity in mind. 
- Don't get anything with depth 4mm or above as they will be hard to fit.
- I had some with dimensions 26x10x3.5 mm and 200mAh capacity which seem to fit below pcb fine or below microcontroller. 

The battery can be placed below PCB, between hotswap sockets and can be secured with some tape.
Placement can be like in the image below (sorry for bad quality).
[image]

You can connect battery as follows: 
- Easiest method is to directly connect battery +- to XIAO pad Bat+,- or nice!nano battery pins bypassing switch.
- Better method is using JST connector B1 (see in ibom) connect battery below PCB then connect BAT+ pin to +pad on XIAO. 
- In case of nice!nano it will automatically get +/- pins just connect the battery via JST at B1 like previous step.

Will add build guide soon, feel free to clarify anything and do check the polarity of battery itself and PCB JST pads.

azhizhinov, 2022-11-07:
Hello.
Thanks for your answers.
I think, there is an issue with B1.
Accordingly to KiCAD, it's S2B-PH with 2.0mm pitch, THT.
But, accordingly to your BOM, it's SM02B-GHS-TB(LF)(SN) with 1.25 pitch, SMD.
Thanks for your time.

snsten, 2022-11-07:
You are correct there is a mismatch, since I used Seeed Fusion Open Parts Library [OPL](https://www.seeedstudio.com/opl.html) to get components.
 I couldn't get the 2mm pitch THT connector there it might be there but couldn't find it on that site.
You can solder the 1.25mm connector directly on the pads not ideal but it will work. 
I would change the part to more suitable version later created an issue #5.

snsten, 2022-11-13:
> Hello. May I ask you about a firmware, intended to use with your keyboard? Is that pretty KLOR firmware? If so, whether this options can be changed from these: `/* encoder support */ #define [...]

Added initial QMK config [here](https://github.com/snsten/qmk_firmware) will add a better keymap and config for RP2040 based boards like KB2040 later.

snsten, 2022-12-24:
PCB with fixed routes works correctly.

closed 2022-12-24 by snsten