snsten/Klein #6: OLED i2c pinout mislabeled or misrouted

idesignstuff, 2022-11-10:
[image]
In my (very limited) experience with PCB design, the square pad = GND and the pinout for a OLED on Amazon appears to have the GND pin in the first position. 
[image]
My hypothesis is that the Silkscreen just needs to match the pinout.

snsten, 2022-11-11:
Don't follow the silkscreen for the old PCB gerbers, for both L and R the order is SDA SCL VCC GND 
Fixed the silkscreen labels in the latest commit, the routing is fine just have to bridge the appropriate jumpers for L/R sides.

On R side the GND will be the square pad
[image]

On L side SDA will be the square pad due to flipping of PCB.
[image]

idesignstuff, 2022-11-11:
Boards were shipped this week, so I'll roll with it.

closed 2022-11-11 by idesignstuff

idesignstuff, 2023-01-20:
So, I'm hoping to get an OLED on the left (peripheral) side. The first rev PCB's I'm using have no VCC voltage on the left side OLED pins, but it's there on the touchpad pins.
As the OLED mounts upside down, the pins are all wrong according to the silk on the left side.
they would need to be in the same arrangement (SDA,SCL,VCC,GND) on the keys side of the left board, just like they are on the keys side of the righthand board. As you have a set of jumpers for each side, this looks possible, but I haven't learned kicad yet to check... 
As a bit of encouragement, I'm typing this on my Klein build!