snsten/Klein #7: Clearance of diodes and choc sockets

AlexBaldwin42, 2022-11-15:
On the outside column while using the SOD123 diodes they interfere with the choc sockets.  Still usable by soldering the diodes offset down about a millimeter.

snsten, 2022-11-15:
Can you check if having diodes and switches on the same side and sockets at the other side of PCB have this issue as shown in the attached image. I have not tested with a complete choc build but I think it should work.

[image]

AlexBaldwin42, 2022-11-16:
[image]
Here is the clearance i was talking about.  

There are no clearance issues with the switches when the diodes are mounted to the opposite side so thats a solution.

snsten, 2022-11-18:
I was aware of this issue that's why I had reversible diodes footprint, keeping the issue open might change the position of the diode in next revision.

P.S. White PCB looks great.

snsten, 2023-01-03:
> Can you check if having diodes and switches on the same side and sockets at the other side of PCB have this issue as shown in the attached image. I have not tested with a complete choc build but I think it should work.
> 
> [image]

Issue with using diode on top was that it interfered with the switchplate in case of choc build.
Fixed the diode placement in the latest commit.

closed 2023-01-03 by snsten