snsten/Klein #8: Debug

LonFas, 2023-01-13:
Good afternoon.
I soldered all for a wired connection, flashed the ProMicro power supply this firmware https://github.com/snsten/Klein-qmk, but any my press the keys doesn't recognized.
Are there any points to check for a closure? May be need connect OLED for debug?

````shell
Jan 14 02:22:08 leviathan.lonfas.lab kernel: usb 1-8: USB disconnect, device number 54
Jan 14 02:22:12 leviathan.lonfas.lab kernel: usb 1-8: new full-speed USB device number 55 using xhci_hcd
Jan 14 02:22:12 leviathan.lonfas.lab kernel: usb 1-8: New USB device found, idVendor=feed, idProduct=0000, bcdDevice= 1.00
Jan 14 02:22:12 leviathan.lonfas.lab kernel: usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jan 14 02:22:12 leviathan.lonfas.lab kernel: usb 1-8: Product: klein
Jan 14 02:22:12 leviathan.lonfas.lab kernel: usb 1-8: Manufacturer: ssh
Jan 14 02:22:12 leviathan.lonfas.lab kernel: input: ssh klein as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:FEED:0000.004E/input/input204
Jan 14 02:22:12 leviathan.lonfas.lab kernel: hid-generic 0003:FEED:0000.004E: input,hidraw2: USB HID v1.11 Keyboard [ssh klein] on usb-0000:00:14.0-8/input0
Jan 14 02:22:12 leviathan.lonfas.lab kernel: input: ssh klein Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:FEED:0000.004F/input/input205
Jan 14 02:22:12 leviathan.lonfas.lab kernel: input: ssh klein System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:FEED:0000.004F/input/input206
Jan 14 02:22:12 leviathan.lonfas.lab kernel: input: ssh klein Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:FEED:0000.004F/input/input207
Jan 14 02:22:12 leviathan.lonfas.lab kernel: input: ssh klein Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:FEED:0000.004F/input/input208
Jan 14 02:22:12 leviathan.lonfas.lab kernel: hid-generic 0003:FEED:0000.004F: input,hidraw3: USB HID v1.11 Mouse [ssh klein] on usb-0000:00:14.0-8/input1
Jan 14 02:22:12 leviathan.lonfas.lab mtp-probe[20915]: checking bus 1, device 55: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8"
Jan 14 02:22:12 leviathan.lonfas.lab mtp-probe[20915]: bus: 1, device: 55 was not an MTP device
Jan 14 02:22:12 leviathan.lonfas.lab systemd-logind[533]: Watching system buttons on /dev/input/event18 (ssh klein Consumer Control)
Jan 14 02:22:12 leviathan.lonfas.lab systemd-logind[533]: Watching system buttons on /dev/input/event17 (ssh klein System Control)
Jan 14 02:22:12 leviathan.lonfas.lab systemd-logind[533]: Watching system buttons on /dev/input/event19 (ssh klein Keyboard)
Jan 14 02:22:12 leviathan.lonfas.lab systemd-logind[533]: Watching system buttons on /dev/input/event15 (ssh klein)
Jan 14 02:22:12 leviathan.lonfas.lab mtp-probe[20942]: checking bus 1, device 55: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8"
Jan 14 02:22:12 leviathan.lonfas.lab mtp-probe[20942]: bus: 1, device: 55 was not an MTP device
```

Do I necessary to flash both controllers? Or is one of connected to a computer sufficient?

---

p.s. I can send photos of soldering

LonFas, 2023-01-14:
Perhaps I made a mistake on the firmware steps, and I need to somewhere add a layer separately, but this is not in the instructions?

My steps:
1. Install QMK
2. Clone https://github.com/snsten/qmk_firmware to ~/ (linux os)
3. Clone https://github.com/snsten/Klein-qmk to ~/qmk_firmware/keyboards/klein
4. Run qmk flash -kb klein -km default
5. Reset pins GND and Reset

snsten, 2023-01-14:
Hi, I looked at the pro micro image you shared it seems to be in wrong orientation. You have to solder the pins so it sits usb port side down like shown in build guide image [here](https://raw.githubusercontent.com/snsten/Klein/main/docs/images/buildguide/Trackpad_Mount.JPG) or like [this](https://romly.com/blog_images/gherkin_socketing_promicro/socketing_promicro-07.jpg).

snsten, 2023-01-14:
>p.s. Do I need to flash the second controller? Why is he needed (except than encode or screen), if everything works through the left board? 

Yes you have to flash both sides, about needing two pro micro you can read the reason at  [qmk docs](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_split_keyboard.md).
> In the case of a wireless connection, do the split appear as two keyboards and work independently? What about TRRS cable between they?

No, in case of wireless you will see only one keyboard the right side as "klein" which you can use to connect to you device PC/Phone etc. The left hand side will not show in bluetooth connections it will connect to the right hand side.
In wireless configuration no TRRS cable is required.

LonFas, 2023-01-14:
> Hi, I looked at the pro micro image you shared it seems to be in wrong orientation. You have to solder the pins so it sits usb port side down like shown in build guide image [...]

> Yes you have to flash both sides, about needing two pro micro you can read the reason at [qmk docs](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_split_keyboard.md).

Hm...Fixed. Status has a bit changed:
Left board lightly blink her leds when i press buttons, but nothing printed. Right side doesn't react. If i connect the usb cable to the right controller, the situation will a mirror.

There is a guess: 

- I use 4-pole TRRS cable. How many pins should a TRRS socket have? My has five. Is there an error here? https://imgur.com/a/tfRg7VP
- Do I need to specify in the firmware which controller is the main and which is auxiliary?

I soldered OLED pins. Change at QMK firmware OLED_ENABLE to yes at rules.mk.  Maybe these photos will help to understand where need to look? https://imgur.com/a/Rfb8nt9

p.s. This is my first DIY keyboard

snsten, 2023-01-15:
> I use 4-pole TRRS cable. How many pins should a TRRS socket have? My has five. Is there an error here? https://imgur.com/a/tfRg7VP

The TRRS jack should have 4 pins for each of Tip, Ring1, Ring2, Sleeve. If you have a multimeter you can check if you get continuity of these four connections between the TRRS jack and cable. Otherwise get the Model: PJ-320A TRRS Jack.

> Do I need to specify in the firmware which controller is the main and which is auxiliary?

No need to specify the main. The right side is main and it's already defined in the firmware.

About the LED's replace them with 1N4148W diodes, you can read the suitability of different diode tested at https://www.40percent.club/2016/07/diodes.html

LonFas, 2023-01-17:
I soldered all recommended parts.  For some reason, right split - is leading. If i connect the cable to the left split - i get a mirror QWERTY layout. Is this ok for the default setup, and i need configure in firmware something like [this](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_split_keyboard.md#handedness-by-define)? 

Found strange behavior. The keyboard doesn't recognize the order of clicks well (if several buttons are pressed with a period somewhere less than half a second, similar when we check new keyboard by pressing). While I suspect the controller board. I think flash firmware to Xiao BLE

snsten, 2023-01-17:
> I soldered all recommended parts. For some reason, right split - is leading. If i connect the cable to the left split - i get a mirror QWERTY layout. Is this ok for the default setup, and i need [...]

It's Ok, by default the main side is right hand which connects to the PC via usb. If you want to use the left side as main you can change that in the firmware as mentioned in the qmk link.

> Found strange behavior. The keyboard doesn't recognize the order of clicks well (if several buttons are pressed with a period somewhere less than half a second, similar when we check new keyboard by pressing). While I suspect the controller board. I think flash firmware to Xiao BLE

Some keys are mapped to multiple functions so the error in order could be due to that.
In the [keymap.c](https://github.com/snsten/Klein-qmk/blob/61e426dce58ce546a99520e84cac6a4a74e651bb/keymaps/default/keymap.c#L27) file some of the keys are defined with [mod-tap](https://docs.qmk.fm/#/mod_tap).
For example: LGUI_T(KC_A)  means A when key is tapped and LGUI/win key when held.

LonFas, 2023-01-17:
Thank you. I will deal with controller and firmware.

closed 2023-01-17 by LonFas