snsten/Klein #11: Support more keys

zonetw, 2023-03-09:
If Klein can support 42 keys will be wonderful.
Don't know if this is possible, just want to know.

snsten, 2023-03-12:
It's possible, I had a prototype with 40 keys. Will need I/O expansion as the Xiao doesn't have enough pins to support all the 42 keys + extra features at once. I will try to add it in v2 of the board.
[image]

rafaelpirolla, 2024-03-05:
Could you drop this one into a branch maybe? It would work fine with pro micro ones... :)

rafaelpirolla, 2024-03-25:
I'm trying to edit it but it's getting hard to find all needed footprints and libraries.

snsten, 2024-03-26:
I'll add the 40 keys version in a new branch.
Agree that finding the needed footprints and libraries can be difficult. Will add custom footprints and libraries used in the project to a separate repo later.

snsten, 2024-03-30:
Added the 40 key version in branch [test-extra-keys](https://github.com/snsten/Klein/tree/test-extra-keys)
Include PCB, case and plates for the new design. 
Haven't tested the new design fully so it might have some issues.

rafaelpirolla, 2024-04-02:
Is the encoder still usable, you think?

snsten, 2024-04-04:
Rotary encoder is unaffected and should work correctly. The rotary encoder switch shares the same connection as one of the extra keys added so both will emit the same key output.