snsten/Klein #12: Downloadable gerber file for online PCB printing services

finex, 2023-04-04:
Hi, would be possible to include a downloadable gerber file (.zip) for online PCB printing service? It would be very useful. Thank you very much.

admilsonmarques, 2023-05-12:
I'd love that as well

suriaribeiro, 2023-06-02:
BOM, CPL and Gerber zip would be fantastic!

MagedEWilliam, 2023-10-12:
Was any one able to generate a good gerbers?
when i generate the files and test it in pcbway viewer, it looks off like it has no holes in the mounting holes

i use default kicad settings when generating the gerbers:
[image]

[image]
https://www.pcbway.com/project/OnlineGerberViewer.html