snsten/Klein #14: Question: why does the wireless version support only OLED *OR* Encoder?

bncpr, 2023-08-02:
Hi, why does the seeeduino xiao ble only support either OLED or Encoder. Can't both of then be connected on the I2C bus? Is it a matter of firmware limitation?

snsten, 2023-08-06:
Reason for not supporting both OLED and Encoder:
1. Due to the limited number of pins in XIAO
2. Both of these can be connected on the same I2C bus, some implications of using this configuration are:
   - The behaviour will depend on how the rotary encoder is designed
   - The encoder could short the line when in rest, check the datasheet for exact behaviour of your encoder
   - It could also interfere with the I2C bus voltage / oled when in use
   - Overall adding two device to same pins will be undefined behaviour in the firmware so I will suggest avoiding it

You can use something like this adapter circuit [i2c-encoder-mini](https://www.duppa.net/shop/i2c-encoder-mini/) safely but you will have to modify and add support for it in the firmware accordingly.

bncpr, 2023-08-06:
I see. Thanks for the answer.

closed 2023-08-06 by bncpr