snsten/Klein #16: Switch plate too tight for MX switches

suriaribeiro, 2023-09-03:
Hi, the switch plate STL does not work with MX switches because its too tight. Do you maybe have an editable version of the STL file or revised version of the switch plate?

suriaribeiro, 2023-09-05:
Managed to create an alternative one, see PR here https://github.com/snsten/Klein/pull/17

snsten, 2023-09-09:
Hi, I am using MX build with the provided STL and didn't found this issue. 
Can you let me know what materials are you using for your 3D print, specific switches you are using or any other info relevant for debugging.

suriaribeiro, 2023-09-09:
Hi, maybe my printer is not doing a decent job. I had to shave a fair bit to fit choc switches. For MX switches it was impossible. For printing the plate I used PLA+ on a Elegoo Neptune 2s. I tried both 0.2 and 0.12 layer heights.

suriaribeiro, 2023-09-09:
obviously I would not be offended if you reject the PR given that no one reported this same issue :)

snsten, 2023-09-10:
Can you try to print any other keyboard's switch plate / case like corne etc. and check if the switch tolerance issue is due to the 3D printer settings or with the file I have in the repo.
If I accept PR without knowing the root cause it will cause confusion for new users on choosing which file to 3D print.